PiSupply/Flick #14: Communication seems good, but no Gesture Data

tomvdb, 2017-06-18:
I've installed the flick hat zero and did the one line install. It seems to be talking to the chip (ie. it shows the versions, etc), but I'm not getting any gesture data when running flick-demo. Am I missing something ?

[image]

francesco-vannini, 2017-06-19:
Hi @tomvdb , indeed that's rather strange. can I get an output of
i2cdetect -y 1
Also do you have another Pi you can test with? I presume you are using a Zero :) do you have a Bx you can test with? Could you also try a fresh, updated Raspbian?

tomvdb, 2017-06-19:
ok, so I tried with a fresh install, updated the packages and installed the flick software. Still no go, same result.

i2cdetect -y 1 gives the following:

[image]

I also tried this in a pi3, same result :(

shawaj, 2017-06-19:
Did you try on the pi 3 with a fresh SD card and fresh version of NOOBS, not the same one as used for pi zero?

Also does it still not work if you use it on the machine itself rather than with putty?

tomvdb, 2017-06-20:
(1) first I flashed the sd in the zero again with a new fresh version of raspbian, did the updates and installed the software. same result
(2) then I took a different sd, flashed a new raspbian on there, did the updates and tried it on the pi3, same result.

I didn't ssh into the pi3, that was straight on the monitor/keyboard setup.

I also tried both the lite and the full raspbian :(

francesco-vannini, 2017-06-20:
I think at this point we might need to look into replacing the board. Did
you buy it from us? If so can you open a support request on
https://www.pi-supply.com/contact-us/

tomvdb, 2017-06-20:
No, I bought it locally in South Africa at the pishop.co.za 

I will open a support ticket with them. Thanks for the help.

shawaj, 2017-06-20:
Guessing the screenshot were taken through SSH as they look like putty?

But in any case good you've tried it direct from Pi 3 terminal.

@francesco-vannini do you think it's worth attempting to re flash the firmware just as a last ditch attempt to fix the board? 

It seems like it must be programmed as firmware data is showing and it would be tested out of the factory but it seems strange it can find the chip but not detect any gestures. Not seen this before!

shawaj, 2017-06-20:
Also @tomvdb do you get any signs of life at all if you hold your hand completely over the device, as close as you can?

Anything in the X, Y, Z field at all?

tomvdb, 2017-06-20:
Thats correct, screenshots were made while using putty ssh'd into the pi zero. I get the same on the terminal though on the pi3, I can take some photos if needed.

no, no signs at all. no matter where I put my hand all the way up to touching it.

I'm willing to try and reflash the firmware? how would I go about doing that?

francesco-vannini, 2017-06-20:
@tomvdb refer to the firmware page https://github.com/PiSupply/Flick/tree/master/hardware/Firmware/update and get the HAT enz file

From within /home/pi/Flick/hardware/Firmware/update
cp ../Flick_Zero_V1.1.enz .
./flickupdate

tomvdb, 2017-06-20:
hey, the firmware update seemed to have done the trick. Its working now on my pi3 :)

[image]

shawaj, 2017-06-20:
Awesome!

shawaj, 2017-06-20:
Glad that worked for you

closed 2017-06-20 by shawaj

gazhay, 2017-06-27:
I've got exactly the same problem. I've got as far as trying to flash the firmware and get

````
    /Flick/firmware/update $ ./flickupdate 
    can't open  gpio xfer direction file
    can't open  pin file
    can't open  gpio xfer direction file
    can't open  gpio xfer direction file
    can't open  pin file
     Loader record_count 109
     lib record_count 222
    Flashing loader.
    gestic_flash_begin 
````
Its been stuck at that for 20 mins

gazhay, 2017-06-27:
Ok, have tried it as sudo, and it seemed to make it flash, after a reboot, still nothing. 

````
    ┌────────────────────────────────────────────────────────────────────────────────────────────────────────────┐
       │ Firmware valid: Yes                                                                                        │
       │ Hardware Revison: 99.128                                                                                   │
       │ Params Start Addr: 0x7300                                                                                  │
       │ Library Loader Version: 19.100                                                                             │
       │ Library Loader Platform: Hillstar                                                                          │
       │ Firmware Start Addr: 0x1000                                                                                │
       │ Firmware Version: 1.3.14;p:HillstarV01;x:                                                                  │
       │ DSP: ID9000r2963;i:B;f:22500;nMsg;s:Rel_1_3_prer1784:NM;                                                   │
       └────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
    
````

shawaj, 2017-06-27:
To anyone seeing this thread after it has been closed.......

The firmware issue is very unlikely. You should not reflash the firmware unless it is recommended by one of us directly to you as you can break the board by doing so. It should only be done after all other tests have been performed and ONLY if one of us requests it.

shawaj, 2018-03-17:
@mike-shock can you please start a new issue with exactly what you're seeing?

lionelvrolijk, 2018-12-06:
guys i need help with mine i accidently run the wrong firmware update