PiSupply/Flick #16: Reopen Issue #14

gazhay, 2017-06-27:
Just in case it gets missed, I'm still having the same issue as #14 

Tried flashing firmware to no end.

francesco-vannini, 2017-06-27:
what does
sudo i2cdetect -y 1
report?

gazhay, 2017-06-27:
The same as the post by tomvdb on issue #14 .

    42

gazhay, 2017-06-27:
Should say I was on a fresh NOOBS Raspbian install on an rpi3

shawaj, 2017-06-27:
@gazhay can you tell us what Flick board you are using and what Pi?

Have you tried starting from scratch again? The firmware issue is very unlikely. You should not reflash the firmware unless it is recommended by one of us directly to you as you can break the board by doing so. It should only be done after all other tests have been performed.

gazhay, 2017-06-27:
Flick large and an rpi3.
I have an pi 2, zero and zero w if it would make a difference?

I started from scratch, but can try again with a fresh sd. Won't be able to for a few days but will report back when I do.

francesco-vannini, 2017-06-28:
I am sure you did but have you connected the board as shown here?
https://www.pi-supply.com/make/flick-quick-start-faq/#flick-large-board

gazhay, 2017-06-28:
Yes, I have wired up as in that diagram. 
FYI, I can control the LEDs via gpio.

gazhay, 2017-06-29:
Still nothing on a brand new card with a new noobs and fully apt updated.

francesco-vannini, 2017-06-29:
OK @gazhay at this point I think you might want to get a replacement. Did you buy directly from us? If so can you get in touch via https://www.pi-supply.com/contact-us/

gazhay, 2017-06-29:
Sadly, I bought it from pi-hut. I'll contact them directly.

francesco-vannini, 2017-06-29:
Please do let us know how this goes.

francesco-vannini, 2017-07-17:
Hi @gazhay did you get a replacement? Are things working ok now?

francesco-vannini, 2017-08-02:
Closing this as no answer provided for 16 days. By all means @gazhay do get in touch if you need to reopen it.

closed 2017-08-02 by francesco-vannini

gazhay, 2017-08-02:
Never got a replacement and vendor now out of stock. If I ever manage to get another one will post a new issue.

shawaj, 2017-08-02:
Have you tried flashing the firmware onto the chip again? That may fix it

gazhay, 2017-08-03:
I did before I sent it back, no change. And then I was rebuked for flashing firmware at all.

shawaj, 2017-08-03:
Pi hut will likely reorder from us pretty soon :-)

Have they got the faulty one back?

gazhay, 2017-08-03:
They do.

shawaj, 2017-08-03:
Ok we'll get in touch with them right now