PiSupply/Flick #20: Flick Demo wont start

DamagedPanda, 2017-08-01:
using the latest Raspbian with a Pi3 all up to date.
the flick script was installed but when i run flick-demo i get this error.

pi@raspberrypi:~ $ flick-demo
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 534, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 472, in <module>
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 296, in _read_msg
UnboundLocalError: local variable 'io_error_count' referenced before assignment
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
pi@raspberrypi:~ $

shawaj, 2017-08-01:
How did you install the Flick package?

And you've definitely got it wired up as in the quick start guide?

DamagedPanda, 2017-08-01:
used this guide for the connection.  https://www.pi-supply.com/make/flick-quick-start-faq/
Install went like this fresh install of Raspbian Twice-
From Terminal
sudo apt-get update && sudo
sudo reboot
curl -sSL https://pisupp.ly/flickcode

also tried 

curl -sSL https://pisupp.ly/flickcode | sudo bash

I can control the LED with the code provided here on GitHub.

shawaj, 2017-08-01:
> make/flick-quick-start-faq/

@francesco-vannini was this the same problem we were having before with
someone where the firmware flash fixed it?

francesco-vannini, 2017-08-02:
@DamagedPanda could you run `sudo i2cdetect -y 1` and provide the output
also provide the output of `cat /boot/config.txt`

The fact that you can control the LEDs seems to suggest your wiring is OK (I also saw the picture you sent John)

@shawaj in the case of the firmware issue the demo should still start but not communicate with the board AFAIK see #14 

Also @DamagedPanda running `curl -sSL https://pisupp.ly/flickcode` as you tried on your first attempt will not install anything, it just downloads the install file which without being piped into `sudo bash` will not get executed. A fresh (not updated nor upgraded) Raspbian with our one-liner should be all you need to use Flick.

DamagedPanda, 2017-08-02:
Fresh install of raspbian
ran 
curl -sSL https://pisupp.ly/flickcode | sudo bash

[http://imgur.com/a/wl9Pz](url)

DamagedPanda, 2017-08-02:
fresh install of raspbian
ran
curl -sSL https://pisupp.ly/flickcode | sudo bash
System rebooted . still have the same issue 

pi@raspberrypi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~ $ 

pi@raspberrypi:~ $ cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxtreadme
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
pi@raspberrypi:~ $

shawaj, 2017-08-02:
Have you definitely got I2C turned on? And you've definitely got it wired
correctly?

Weird that it's not showing up on the I2C bus. Never seen that before.

francesco-vannini, 2017-08-03:
@DamagedPanda could you try using a different cable to connect the Flick to the Pi? You could use a 6 wire one as there is no need to connect the LEDs. At this point I'd like to make sure that at least we can see the 0x42 address is shown when running the i2cdetect.

francesco-vannini, 2017-08-29:
Hi @DamagedPanda has the issue been solved?

francesco-vannini, 2017-09-01:
Closing this as no progress on the resolution from requester.

closed 2017-09-01 by francesco-vannini

theold190, 2017-09-01:
I had similar outputs in the log with my Flick (due to faulty HW connections). It was confusing to see all those errors.

Would it be possible to print instead "Flick board was not detected, make sure it is connected properly"? That request is valid for all demos in this repository.

shawaj, 2017-09-01:
@theold90 yes this could be a good idea.

@tvoverbeek @francesco-vannini what do you think to this? Maybe a good way
to deal with these errors in a more "useful" way?