PiSupply/Flick #27: Did not receive firmware info

MichaelBitard, 2017-11-11:
Hi,

Not sure if this is the right place to post this kind of issue.

I have set up a rpi with a flick large.

The RPI:
```
Raspberry Pi Details:
  Type: Model B, Revision: 02, Memory: 512MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi Model B Rev 2
  * This Raspberry Pi supports user-level GPIO access.
```
The sudo i2cdetect -y 1
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     
```

I'm able to change the LED according to your readme with 
```
gpio -g mode 22 out
gpio -g write 22 1
This will turn off the red LED

gpio -g mode 4 out
gpio -g write 4 0
```

But whenever I launch the flick-demo, I get:

```
Did not receive firmware info
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    import flicklib
  File "/usr/local/lib/python2.7/dist-packages/flicklib.py", line 479, in <module>
    sys.exit()
NameError: name 'sys' is not defined
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/local/lib/python2.7/dist-packages/flicklib.py", line 462, in _exit
    _stop_poll()
TypeError: 'NoneType' object is not callable
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/local/lib/python2.7/dist-packages/flicklib.py", line 462, in _exit
    _stop_poll()
TypeError: 'NoneType' object is not callable
```

I'm using a freshly installed 2017-09-07-raspbian-stretch-lite

I tried with python2 and 3, still the same error.

Any idea on something I did wrong?
Thanks

shawaj, 2017-11-12:
Just to clarify this is on an original 26 pin RasPi?

MichaelBitard, 2017-11-12:
Yes, It's a model B rev2 with 26 pin

I will try it on a RPI 3 later, if you think this may be the issue.

tvoverbeek, 2017-11-12:
SInce i2cdetect shows the i2c communication works, my guess is you have exchanged the TS (transfer status) and RESET connections (pins 11 and 13, GPIO 17 and 27).
Just checked with my Flick Large and if TS is not connected correctly I get the same error.
See also the pictures of the Flick Large connections on https://www.pi-supply.com/make/flick-quick-start-faq/

MichaelBitard, 2017-11-12:
I checked that, tried to switch the 11 and 13 but still nothing

MichaelBitard, 2017-11-12:
Mmmh, I'm not able to have `i2cdetect` to detect the flick on the i2c bus anymore :(

MichaelBitard, 2017-11-17:
Still no idea on this one? I tried on 2 different rpi B rev2, should I try on rpi 3?

shawaj, 2017-11-17:
The original Pi uses a different i2c bus I believe?

Have you edited the code to account for that?

Can you try on a Pi 3 please yes? Then we can rule out an error with the
Flick board.

tvoverbeek, 2017-11-18:
> The original Pi uses a different i2c bus I believe?
Only the B Rev1 has i2c0.
All others have i2c1.
@MichaelBitard uses a B Rev2, so code changes should not be needed.

closed 2017-11-18 by tvoverbeek

MichaelBitard, 2017-11-27:
I just tested with a pi3, and I have the same issue "Did not receive firmware info"

Any idea on what to do next?

shawaj, 2017-11-28:
Try i2cdetect -y 1

Paste screenshot of output here

Thanks

MichaelBitard, 2017-11-28:
That's what I did in the description of this issue:

The sudo i2cdetect -y 1
```bash
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     
```

francesco-vannini, 2017-11-28:
Could you try flashing the firmware again by following the procedure described here
https://github.com/PiSupply/Flick/blob/master/firmware/update/README.md

MichaelBitard, 2017-11-28:
I tried that, and I got this:

```bash
cd /home/pi/Flick/firmware/update
sudo cp ../Flick_Large_V1.1.enz .
sudo ./flickupdate Flick_Large_V1.1.enz

can't export xfer pin
can't export reset pin
can't open  gpio xfer direction file
can't open  pin file
 Loader record_count 109
 lib record_count 222
Flashing loader.
gestic_flash_begin 
read error
ReceiveMsg 
 
write error
write error
gestic_flash_write: failed 
write error
gestic_flash_write: failed 
write error
gestic_flash_write: failed 
write error
gestic_flash_write: failed 
write error
gestic_flash_write: failed 
write error
gestic_flash_write: failed 
write error
gestic_flash_write: failed 
write error
...
...
...
estic_flash_write: failed 
write error
gestic_flash_end ret -1 
write error
Flashing library.
gestic_flash_begin 
read error
ReceiveMsg 
 
write error
write error
gestic_flash_write: failed 
...
...
...
write error
gestic_flash_write: failed 
write error
gestic_flash_write: failed 
write error
gestic_flash_end ret -1 
write error
```

francesco-vannini, 2017-11-28:
Are you using a freshly created SD card with update Raspbian on it? Please let's move this to our support sale@pi-supply.com as this is not a problem with the code base and we can get pictures of your setup.