PiSupply/Flick #28: README.md outdated or incorrect?

srcshelton, 2017-12-29:
I was about to return my Flick as faulty, as one of the early instructions is to run:

`sudo i2cdetect -y 1`

(noting the 'Tpo' typo just above) to confirm that the hardware is working.  On my RPi3B with HAT Flick board, I enabled i2c, rebooted, yet the above command still returns no devices found on any i2c address.

*However* the included demo does provide X, Y, and Z co-ordinates and I've got it to report east-west and west-east swipes after a /very/ brief try.

So is i2c no longer required?  Does it only pertain to the larger external board?  OTOH, should i2c be working and do I lose any functionality if it isn't?

What GPIOs does the HAT make use of?

srcshelton, 2017-12-29:
The `flicklib.py` library code performs i2c reads and writes to address `0x42` hard-coded, without any discovery - so is it just the case that the current firmware isn't responding as necessary for `i2cdetect` to show anything, but the ability to read device data and co-ordinates via the demo code is canonical proof of correct operation?

shawaj, 2017-12-29:
It should be detected on the i2c bus.

What do you see when you do i2cdetect command? Can you send a screenshot or paste the output here?

shawaj, 2017-12-29:
But yes, it's obviously working if it's reporting data back to the software. The interesting thing is why it isn't showing on the i2c bus when it is clearly there!

srcshelton, 2017-12-29:
Confirmed - the Flick device **does** show in `i2cdetect`, but only *after* it's already been used!
The initialisation performed by `flicklib.py` causes the Flick device to appear on i2c, but only once it's been run!  This is potentially confusing for new users...

shawaj, 2017-12-29:
@tvoverbeek @francesco-vannini have you seen similar behaviour with these boards?

srcshelton, 2017-12-29:
```
pi@piflick:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@piflick:~ $ src/flick/bin/flick-demo 
Traceback (most recent call last):
  File "src/flick/bin/flick-demo", line 147, in <module>
    wrapper(main)
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "src/flick/bin/flick-demo", line 145, in main
    time.sleep(0.1)
KeyboardInterrupt
pi@piflick:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

It's not a huge problem, of course - but the documentation should probably be updated to reflect this, so that more people don't do the i2c test as the first post-hardware-installation check, and then worry they they've broken something during installation!

shawaj, 2017-12-29:
thanks @srcshelton that is great - we will investigate further with some new boards we have here and try to replicate.

do you have more than one Pi by any chance? I would be interested to know if you moved the board that is detected by one pi, to another pi and tried the same thing whether it would still occur on that "fresh" pi?

tvoverbeek, 2017-12-30:
Running `i2cdetect -y 1` (sudo not needed any more) directly after boot gives indeed no reply at address 0x42.
This is because the MGC3130 chip is in deep sleep after boot.
You have to reset the chip first before i2c works with Flick.
This is why it works after running any flick demo program.
You can send the reset (low pulse on GPIO 17) manually:
For example if you have the gpio program installed (package wiringpi) the following one liner works directly after boot:
```
gpio -g mode 17 out; gpio -g write 17 0; gpio -g write 17 1; i2cdetect -y 1
```

tvoverbeek, 2017-12-30:
Updated README.md accordingly.
@srcshelton Stuart, thanks again for reporting this.
Closing issue

closed 2017-12-30 by tvoverbeek

shawaj, 2017-12-30:
thanks @tvoverbeek @srcshelton :-)