PiSupply/Flick #29: Demo code can't detect touch/tap/double-tap gestures with official Flick case

srcshelton, 2017-12-29:
The official Flick case, used closed, increases the minimum separation between the Flick HAT and the user's finger, and so the demo code isn't able to detect touch, tap, or double-tap events.

It would be helpful to add an on-screen notice and a simple user-editable flag to the demo code to account for this separation and allow the demos to fully work when the official case is used.

shawaj, 2017-12-29:
Thanks for posting this :-)

We have actually been thinking about a way to solve this for a while. Think it may require a firmware update because actually the gesture detection is hard coded into the MGC3130 chip. I am not 100% sure whether you can alter how these function.

We will investigate. Adding in @francesco-vannini and @tvoverbeek to see if they have any wise ideas (they usually have more than me!!)

checkerap, 2018-12-02:
any updates on this, it's been a year and I just picked up a new Flick board, same thing happens. tap, double tap are not working when the board is covered, tap it is somewhat works without a cover, double tap not that much, very unreliable/useless.

shawaj, 2018-12-02:
If you need to use specific functions more accurately, the best way is to build a custom GestIC firmware yourself, deactivating the gestures you don't need. This makes the ones you do need more accurate.

This is not a bug but just the nature of a chip like this.

closed 2018-12-02 by shawaj

tvoverbeek, 2018-12-02:
Just wondering if when using a 3B or 3B+ the on-board WiFi and Bluetooth disturb the electric field used by the Flick to detect gestures, taps, etc.

checkerap, 2018-12-05:
@shawaj understand the limitations, I may have misread the ads, I thought all gestures except touch are "3d air-based". I was hopping to use the tap as an "enter/click" type event. would be great if the firmware can be adjusted.