PiSupply/Flick #30: flick-volctrl not working

gabrielduhoux, 2018-01-11:
I'm having trouble with the flick-volctrl functionality. 
When I run it, the response in LXTerminal is: Compatible audio card not found".
I tried the same using a USB Sound-Card and a HDMI Display-Controller but the response is still the same. 
Now I have a speaker connected to the aux exit on the Raspberry Pi 3 b. The firmware is up to date and the flick-demo shows the Airwheel working as expected.
Of course I have the Alsa Audio Project installed and working.
Any help on that matter would be appreciated – thanks! :)

francesco-vannini, 2018-01-12:
Hi, @gabrielduhoux , I have just tested things with a fresh Raspbian and a RPi 3B all looks good to me.

could you post your config.txt file
`cat /boot/config.txt`
and an ouptput of
`aplay-l`

gabrielduhoux, 2018-01-12:
```
~ $ cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# NOOBS Auto-generated Settings:
hdmi_force_hotplug=1
start_x=0
gpu_mem=128

display_rotate=3

dtparam=i2c=on
dtparam=i2c_arm=on
```

aplay-1 --> `Command not found`

tvoverbeek, 2018-01-12:
Note the command should be `aplay -l` (letter lowercase L not the digit 1)
Here is the output on my RPi 3B:
```
pi@rpi-stretch-full:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@rpi-stretch-full:~ $ 
```

gabrielduhoux, 2018-01-12:
Alright, thanks I missed that one @tvoverbeek.
Got the same output as you now.

But running flick-volctrl still getting `Compatible audio card not found`. 
Any ideas on how to fix that?

shawaj, 2018-01-12:
@francesco-vannini looking at the code it is only set up to control JustBoom DAC / Amp cards not just the standard onboard audio or the Digi cards.

Perhaps we should extend the script to cater for those as well, albeit using the software mixer I guess? What do you think?

tvoverbeek, 2018-01-12:
Problem is in the last commit by @francesco-vannini in the set up the mixer section of main():
```
# Setup the mixer
    for i in range(len(alsaaudio.cards())):  # Finds the JustBoom card or default to onboard
        if (alsaaudio.cards()[i] == 'sndrpiboomberry' or alsaaudio.cards()[i] == 'sndrpijustboomd'):
            card = 'JustBoom DAC'
            control = 'Digital'
            cardId = i
            break
        else:
====> card = 'bcm2835 ALSA' <=====
            control = 'PCM'
            cardId = i
    try:
        mixer = alsaaudio.Mixer(control=control, cardindex=cardId)
    except:
        exit("Compatible audio card not found")
```
.Name is not `bcm2835 ALSA` but just `ALSA`
@gabrielduhoux suggest you copy the flick-volctrl source from github to your own directory and replace 'bcm2835 ALSA' with just 'ALSA' and try again.
@francesco-vannini can you test this and fix the repository?

tvoverbeek, 2018-01-12:
Thinking a bit more, the name change is not needed since you only use control and cardindex in setting the mixer.
However if there is an other sound card installed (not sndrpiboomberry or sndrpijustboomd) then you will end up with control='PCM' and cardId > 0.
The internal audio is card 0 -> mixer setup fails.
So I suggest the following instead:
```
# Setup the mixer
    for i in range(len(alsaaudio.cards())):  # Finds the JustBoom card or default to onboard
        if (alsaaudio.cards()[i] == 'sndrpiboomberry' or alsaaudio.cards()[i] == 'sndrpijustboomd'):
            card = 'JustBoom DAC'
            control = 'Digital'
            cardId = i
            break
        elif alsaaudio.cards()[i] == 'ALSA':
            card = 'bcm2835 ALSA'
            control = 'PCM'
            cardId = i
            break

    try:
        mixer = alsaaudio.Mixer(control=control, cardindex=cardId)
    except:
        exit("Compatible audio card not found")
```

shawaj, 2018-01-12:
Thanks Ton, as always :-)

tvoverbeek, 2018-01-13:
New day, new logic.
The code above will select the built-in ALSA card if both cards are enabled since the built-in card is usually card 0.
The following code first scans all cards to find the JustBoom one.
If not found it scans the cards again to find the built-in 'ALSA' card.
```
    # Setup the mixer
    cardId = -1
    for i in range(len(alsaaudio.cards())):  # Find the JustBoom card
        if (alsaaudio.cards()[i] == 'sndrpiboomberry' or alsaaudio.cards()[i] == 'sndrpijustboomd'):
            card = 'JustBoom DAC'
            control = 'Digital'
            cardId = i
            break
    if cardId == -1:                             # JustBoom card not found
        for i in range(len(alsaaudio.cards())):  # Find the default onboard ALSA card
            if (alsaaudio.cards()[i] == 'ALSA'):
                card = 'bcm2835 ALSA'
                control = 'PCM'
                cardId = i
                break
    if cardId == -1:
        exit("Compatible audio card not found")
    try:
        mixer = alsaaudio.Mixer(control=control, cardindex=cardId)
    except:
        exit("Compatible audio card not found")
```

shawaj, 2018-01-13:
If the JustBoom Digi is attached, will this allow software control of the volume as well?

tvoverbeek, 2018-01-13:
flick-volctrl jut commands the mixer volume.
Any other way to command the mixer volume (i.e. the usual volume slider on
the desktop) will work

shawaj, 2018-01-13:
Ah ok so for the DAC driver it's controlling hardware mixer directly but for on board audio it's just controlling the software mixer? Or the onboard audio has a hardware mixer?

Because with Digi hat, in normal operation, the normal software volume slider on taskbar doesn't work does it @francesco-vannini ?

francesco-vannini, 2018-01-14:
@tvoverbeek thanks for this, I'll test and amend the code. Strange thought that as I tested the whole thing the other day everything was working for me as I said above and I could control the bcm2835 volume so what the user is experiencing must anyway be related to something else.
As for the Digis @shawaj  I had created this ALSA configuration https://github.com/PiSupply/JustBoom#digi-boards when we introduced the rotary encoder article which creates a virtual volume control. @gabrielduhoux do things work for you now?

gabrielduhoux, 2018-01-14:
@tvoverbeek I did what you recommended – still not finding a compatible audio card. 
If I buy that JustBoom Digi (https://www.justboom.co/product/justboom-digi-hat/), would that help instead of using the Aux exit of the Pi?

tvoverbeek, 2018-01-14:
With no other audio cards connected even the original code should detect the built-in ALSA card as the only audio card.
Are you still getting 'Compatible audio card not found'?
Since your 'aplay -l' lists the ALSA audio card, something else must be going on.

You can still get card detected and no audio out of the 3.5 mm jack if the audio is routed to HDMI.
You can force it to the 3.5mm jack e.g. by running 'sudo raspi-config' and selecting Advanced->audio.
I would first try to get the built-in audio detected before considering a JustBoom hat (although PiSupply would not mind if you did ;-))

An other thing you could try is running Python interactively to see what cards alsaaudio sees:
Here is my output with no other audio cardsa/HATs connected:
```
pi@rpi-stretch-full:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import alsaaudio
>>> alsaaudio.cards()
[u'ALSA']
```

gabrielduhoux, 2018-01-14:
My audio is routed to the 3.5 mm jack, when playing music, the audio works fine. 
But the `flick-volctrl` still returns `Compatible audio card not found`.

Running `python` gives me:
`pi@raspberrypi:~ $ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import alsaaudio
>>> alsaaudio.cards()
[u'ALSA', u'Device']`

tvoverbeek, 2018-01-14:
Yes, until the repo gets updated you can use the attached version of flick-volctrl (Attached file flic-volctr.py.txt)
Let me know if this version works with your usb mike attached.
[flick-volctrl.py.txt](https://github.com/PiSupply/Flick/files/1629615/flick-volctrl.py.txt)

gabrielduhoux, 2018-01-14:
Thanks for the file, but it does not work unfortunately `Compatible audio card not found` again...

tvoverbeek, 2018-01-14:
Just tested it (by forcing the cards() to return a 2 element list [u'ALSA', u'Device'])
and it detected the ALSA card properly.
You need to run the file directly, i.e. `python flick-volctrl.py.txt` (if you did not rename it)

gabrielduhoux, 2018-01-14:
Doing that I get a syntax error: 
`python flick-volctrl
  File "flick-volctrl", line 74
    fwwin.erase()
        ^
SyntaxError: invalid syntax`

shawaj, 2018-01-14:
@francesco-vannini if you've set the Digi up like on that repo, will this script work with it? Or does it need different calls? Would be good to support it and maybe put a comment in the code for where to find the setup instructions on the JustBoom repo?

tvoverbeek, 2018-01-14:
@gabrielduhoux My bad. In line 71 a closing parenthesis is missing.
`datawin = curses.newwin(4, curses.COLS - 6, 4, 3` should be
`datawin = curses.newwin(4, curses.COLS - 6, 4, 3)`
Could you add the closing parenthesis and try again (last time, I hope).

gabrielduhoux, 2018-01-14:
@tvoverbeek perfect!! Thanks a lot! I didn't see that one. 

@tvoverbeek @francesco-vannini and everyone: Thank you for all the help!! working perfectly now :) 👍

tvoverbeek, 2018-01-14:
Change commited to master.

closed 2018-01-14 by tvoverbeek

francesco-vannini, 2018-01-15:
@tvoverbeek thanks for committing the new code

shawaj, 2018-01-15:
@francesco-vannini did you see my question?

francesco-vannini, 2018-01-15:
yes (I put a thumbs up) don't worry @shawaj I wrote this down and will work on it

shawaj, 2018-01-15:
Sorry, didn't get notified about the thumbs up