PiSupply/Flick #31: Flick demo/s not detecting gestures

matthew1809, 2018-01-17:
Installed Board and verified w/ `i2cdetect -y 1`
- Correctly showing 42 as address

Ran `flick-demo`
- Verified valid firmware etc.

Began gesturing over board (no cover)
- No response when running `flick-demo`, or `flick-snail`

Wondering how to debug further, apologies for any naivety, beginner speaking!

matthew1809, 2018-01-17:
Seems to be similar to https://github.com/PiSupply/Flick/issues/14

matthew1809, 2018-01-17:
Nothing logging through demo or snail.

shawaj, 2018-01-17:
Which board have you got?

matthew1809, 2018-01-18:
https://uk.rs-online.com/web/p/human-machine-interface-hmi-development-kits/1365431/

The Pi 3 is model B

shawaj, 2018-01-18:
Have you tried on any other pis?

matthew1809, 2018-01-18:
Don't have any other Pi's to try on I'm afraid

matthew1809, 2018-01-18:
Any other method of debugging?

francesco-vannini, 2018-01-18:
So nothing else is connected to your Pi, you are running a fresh version of Raspbian and the Flick board is correctly wired to your RPi 3B?
BTW @shawaj was asking which Flick you have not which Pi

tvoverbeek, 2018-01-18:
The link @matthew1809 gave (https://uk.rs-online.com/web/p/human-machine-interface-hmi-development-kits/1365431/) shows the Flick HAT.

matthew1809, 2018-01-18:
As @tvoverbeek said 👍 

So nothing else is connected to your Pi
- usb keyboard and mouse, hmdi for display

you are running a fresh version of Raspbian
- correct

the Flick board is correctly wired to your RPi 3B?
- As per the following from the guides, yes ` Place the Flick HAT on top of the Raspberry Pi by gently pushing the female header onto the Raspberry Pi male header.`

francesco-vannini, 2018-01-18:
Sorry I hadn't seen the link. I am out of options then, there could be grounds to return the item then.

matthew1809, 2018-01-18:
No worries!

Is there a layer between the demo and the raw recognition of the gesture, where we could inject some logging?

Or, is it clear cut that if not logging in the demo, it's not being picked up by the board?

shawaj, 2018-01-18:
Firmware refresh is the best option really now. It might just be corrupted somehow

If no success with that then we will replace

mike-shock, 2018-03-14:
I have the same problem as **matthew1809** with both of my Flick HAT and Flick Zero pHAT which I bought from ModMyPi.com. I tried them with 2 Raspberry Pi Zero W and with Raspberry Pi 3 Model B - no detection of gestures, only hardware info's shown on the screen. Both boards have Revision Number 99.128.
**shawaj** in Issue#14 said that reflashing of the firmware is not recommended.
What shoud I do now? Return the sensor boards to the seller?

shawaj, 2018-03-15:
@mike-shock have you tried the firmware refresh?

mike-shock, 2018-03-15:
No. I may try if you recommend to do it.

shawaj, 2018-03-17:
@mike-shock can you start a new issue with what you're seeing?