PiSupply/Flick #33: No gesture data ... but can turn on/off leds

VolkerH, 2018-01-25:
Hi,

I seem to have a similar problem as here: https://github.com/PiSupply/Flick/issues/14

No gesture info with any of the demo programs.

* Brand-new flick large ordered through RS Components in Germany
* Fresh install of Raspbian (with latest noobs) on Pi 3B. 
* Installed flick from github using curl as in Readme.md
* Triple check board is wired up correctly
* I2C is enabled.
* I can turn the red / green LEDs on and off using the GPIO commands.
* board is detected

```
pi@raspberrypi:~/Flick $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~/Flick $ ^C
```

flick-demo shows 
```
                             **** Flick Demo ****

   ┌────────────────────────────────────────────────────────────────────────┐
   │ X Y Z     :                                                            │
   │ Flick     :                                                            │
   │ Airwheel  :                                                            │
   │ Touch     :                                                            │
   │ Tap       :                                                            │
   │ Doubletap :                                                            │
   └────────────────────────────────────────────────────────────────────────┘

   ┌────────────────────────────────────────────────────────────────────────┐
   │ Firmware valid: Yes                                                    │
   │ Hardware Revison: 99.128                                               │
   │ Params Start Addr: 0x7300                                              │
   │ Library Loader Version: 19.100                                         │
   │ Library Loader Platform: Hillstar                                      │
   │ Firmware Start Addr: 0x1000                                            │
   │ Firmware Version: 1.3.14;p:HillstarV01;x:                              │
   │ DSP: ID9000r2963;i:B;f:22500;nMsg;s:Rel_1_3_prer1784:NM;               │
   └────────────────────────────────────────────────────────────────────────┘

                               Control-C to exit
```

Just in case 

```
pi@raspberrypi:~/Flick $ cat /boot/config.txt 
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# NOOBS Auto-generated Settings:
hdmi_force_hotplug=1
pi@raspberrypi:~/Flick $ 

```

What should I do ? 
Try and flash firmware as suggested to poster in issue 14? 
Return the board to RS Components ?

Thanks

VolkerH, 2018-01-25:
Continued to work on this. 
I cross-checked with another Raspberry Pi 3b (brand new out of the box). Same issue.
As I am impatient I tried the firmware update. Firmware update ran without error, however no change, still not getting any gestures.

shawaj, 2018-01-25:
Could you send some pictures of both sides of the board, and high
resolution as you can?

And also of the connections?

Thanks

VolkerH, 2018-01-25:
Hi 
Thanks for your reply.
I took some pictures and uploaded them to a flickr album https://www.flickr.com/gp/94122950@N05/1f22zH

VolkerH, 2018-01-26:
Ok. I was testing at home yesterday. At work I found out that my colleague had ordered a second flick large, so he could experiment as well. 

The second board works, I get meaningful data out of flick-demo.

I guess I should exchange the defective board via RS components ?

shawaj, 2018-01-26:
Could you try with your cable and the other board to see if the cable is defective?

And could you try removing the plastic part in top of the header pins on the Flick blocking the cables from getting a proper connection?

That's only there to make it easier for the pick and place machine to place the component during manufacture

VolkerH, 2018-01-27:
I did check the working board with the cable that came with the (suspected) defective board. 
The new board works with both cables.  The old board works with neither cable.

I will try and check whether the plastic part makes a difference. I highly doubt it.
The connections are mainly I2C bus pins GND, VCC and the two LEDs as I see it. 

If the I2C bus pins didn't work I don't think the demo program would be able to communicate with the board to display Firmware versions etc.

shawaj, 2018-01-27:
It's extremely unlikely that the i2c would work but the gestures not.

Perhaps try with shorter jumper cables if you can. Perhaps the capacitance in the cable is too high.

We've never seen this issue with the zero or hat boards so I can't think what else it could be frankly...

mmilann, 2018-01-27:
There is c++ test program that can be used alternatively to python for
testing:
https://github.com/PiSupply/Flick/tree/master/firmware/libcpp

Readme explains how to build and also requires to install wiringpi:
sudo apt-get install wiringpi

tvoverbeek, 2018-01-27:
@mmilann The c++ test program does not produce any output because no outputs are enabled.
See #34

mmilann, 2018-01-27:
It will work if it was not previously locked by another application,
because by default after power-up gesture data in payload are not locked.

tvoverbeek, 2018-01-27:
I did not get any output when starting from power-off and not running any other flick software before.
As far as I can tell you need to lock the outputs, otherwise the offsets for the various fields in the sensor data message are not constant.

VolkerH, 2018-01-28:
Hi, 
thanks for the suggestions. I've been quite busy this weekend developing something cool with the working board. I don't have shorter jumper cables at hand.

If the capacity of the cable was the problem I would suspect that both flick boards would have a problem. However, the second board I tried is rock-solid. No glitches with either cable.

Will try clipping off the extra plastic bit for the picking machine and see whether it works. 
Not sure when I will get to it though, have quite a busy week ahead. 

How does the firmware flash update work? Doesn't it work over I2C? It ran without error message, therefore I came to the conclusion that I2C must be working.

shawaj, 2018-01-28:
Did you try the c++ test program?

VolkerH, 2018-01-28:
No, I didn't yet. I left the defective board at work so no chance to test
on the weekend.

shawaj, 2018-01-29:
Ok let us know when you've managed to test

wit009, 2018-03-27:
Hi Guys,

I have exactly the same behaviour as VolkerH - any suggestions?

VolkerH, 2018-03-27:
Completely forgot about this ticket. When I was working on this I had an urgent deadline. Gave back the flick that wasn't working to our purchasing department and used another one. I don't think they returned it to RS as it probably wan't worth their time and the shipping cost to get the money back.

Never tried the C++ program. IMHO if a flick doesn't work out of the box with the flick-demo program I consider the device broken. Wouldn't be much use anyway if it turns out to work with the C++ program when I want to use it via the python API.
With a working flick device flick-demo works just fine. As I also tested a number of different cables so I'm fairly sure it wasn't a bad connection.

Set up a nice interactive display with the working flick, using a Raspberry pi to drive a 4K UHD monitor (it's possible). I really like the flick for this when it works. This is now already installed ... I just received another flick board this week to continue development (can't use the one in the installation).

shawaj, 2018-03-27:
@VolkerH it is unlikely to be broken. At worst it probably needs a firmware update. You can do this very easily. So I would recommend getting it back from purchasing and seeing if you can fix it, if it is not too late :-)

shawaj, 2018-03-27:
@VolkerH or send it back to us, as it would be nice to see if we can actually work out why this is occuring

VolkerH, 2018-03-27:
Well, I did run the firmware update as I mentioned in my second post. Firmware update ran without error but didn't change the fact that there was no gesture sensor data.

I need to see whether I can get it back or whether they discarded it already.

ChristopherRush, 2018-09-24:
@VolkerH @shawaj I'm closing this since we don't have the faulty board to test and there wasn't a resolution.

closed 2018-09-24 by ChristopherRush