PiSupply/Flick #34: c++ flick_example does not work

tvoverbeek, 2018-01-27:
There is one statement missing in flick_example.cpp:
```
        usleep(250000);
        int ret = flick.SetRuntimeParameter(0xa1, 0x1f, 0xFFFFFFFF); // lock data output   <-----
        ret = flick.SetRuntimeParameter(0x80, 0x1b, 0x1F); // disable calibration
        //printf("ret: %d \n", ret);
```
Enabling the output makes it work.

shawaj, 2018-01-27:
Thanks @tvoverbeek - is that the only issue with it? 

Which line is this on so can easily find it to fix?

tvoverbeek, 2018-01-27:
Yes, rest is OK.
The extra line has to be added after line 117 in flick_example.cpp

shawaj, 2018-01-27:
Fixed now - https://github.com/PiSupply/Flick/blob/master/firmware/libcpp/flick_example.cpp

Does that look ok @tvoverbeek ?

Guess we need to upload a new compiled version as well, once you are happy with it

tvoverbeek, 2018-01-27:
OK

shawaj, 2018-01-27:
Have you got a compiled version you could upload? If not no worries, I'll sort it when I'm next by a Pi

tvoverbeek, 2018-01-27:
Compiled version uploaded

closed 2018-01-27 by tvoverbeek