PiSupply/Flick #36: Problem after update Firmware

stemeh, 2018-03-13:
I have update the Flick HAT and now wenn I start the Flick-demo The Raspberry show this Error :
Traceback (most recent call last):
File "/usr/local/bin/flick-demo", line 4, in <module>
__import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 534, in run_script
self.require(requires)[0].run_script(script_name, ns)
File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1445, in run_script
exec(script_code, namespace, namespace)
File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 4, in <module>
__import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
File "build/bdist.linux-armv7l/egg/flicklib.py", line 480, in <module>
File "build/bdist.linux-armv7l/egg/flicklib.py", linTraceback (most recent call last):
File "/usr/local/bin/flick-demo", line 4, in <module>
__import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 534, in run_script
self.require(requires)[0].run_script(script_name, ns)
File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1445, in run_script
exec(script_code, namespace, namespace)
File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 4, in <module>
__import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
File "build/bdist.linux-armv7l/egg/flicklib.py", line 480, in <module>
File "build/bdist.linux-armv7l/egg/flicklib.py", line 278, in _handle_firmware_info
Exception: An invalid GestiIC Library was stored, or the last update failed
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
Error in sys.exitfunc:
Traceback (most recent call last):
File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
e 278, in _handle_firmware_info
Exception: An invalid GestiIC Library was stored, or the last update failed
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
Error in sys.exitfunc:
Traceback (most recent call last):
File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
func(*targs, **kargs)
File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
Can you tell me how I can solve this problem ?

stemeh, 2018-03-13:
no the update go without problems i retry it more times.

This ist the info from Raspberry :  

Flick_HAT_V1.1.enz  flickupdate  flickupdate.c  Library.c  Loader.c  README.md
pi@pi-top:~/Downloads/Flick-master/firmware/update $ ./flickupdate Flick_HAT_V1.1.enz
can't open  gpio xfer direction file
can't open  pin file
can't open  gpio xfer direction file
can't open  gpio xfer direction file
can't open  pin file
 Loader record_count 109
 lib record_count 222
Flashing loader.
gestic_flash_begin 
ReceiveMsg 

c��d 1.3.14;p:HillstarV01;x:        ;DSP:ID9000r2963;i:B;f:22500;nMsg;s:Rel_1_3_prer1784:NM; 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
[... 261 lines cut ...]
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_write: success 
gestic_flash_end ret 0

ChristopherRush, 2018-03-14:
Have you tried re-installing the Flick Repository?

stemeh, 2018-03-14:
Yes, I also downloaded it again but it is the same problem

stemeh, 2018-03-14:
Now I have use a new installation from rasperryPI strech an make update and upgrade and then install the Flick git.

and the same problem

 flick-demo
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 739, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1501, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 480, in <module>
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 278, in _handle_firmware_info
Exception: An invalid GestiIC Library was stored, or the last update failed
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable

stemeh, 2018-03-14:
Now I Have make the update from Flick again now the Porblem is fixed . Thank you for help !

ChristopherRush, 2018-03-14:
To anyone seeing this thread after it has been closed.......

You should not reflash the firmware unless it is recommended by one of us directly to you as you can break the board by doing so. It should only be done after all other tests have been performed and ONLY if one of us requests it.

closed 2018-03-14 by ChristopherRush