PiSupply/Flick #38: Flick gesture stopped working(south - north etc)

donmaronio, 2018-06-19:
Hi, i've installed flick few days ago and averything was working great, but i've added to my system Allo Boss DAC and S.usv ups basic yesterday, and since then flick does not work properly. What i mean is that taps double taps and touch is working fine, flick-demo shows position of my finger above, but flicking does not, once for a while demo will show that i've flicked over it but very rearly. Do you know what it could be?

tvoverbeek, 2018-06-20:
Flick is dependent on detecting disturbances in its electric field caused by your hand.
Mu guess is that one (most likely the voltage converters on the UPS) or both HATs disturb the electric field too much to get a good detection of flicks.
Do you have the DAC, UPS and Flick all stacked on top of each other?

donmaronio, 2018-06-20:
I've got UPS plugged on to the rpi and on that i've got gpio expander (https://botland.com.pl/26910-large_default/ekspander-wyprowadzen-gpio-hat-nakladka-dla-raspberry-pi-32b.jpg) and to that expander i've got plugged flick and dac

ChristopherRush, 2018-09-24:
@donmaronio Any luck on this?

666github666, 2019-02-21:
Hi, I have similar setup and same issue. Is there any way how to limit electric field disturbance by other components on Flick?