PiSupply/Flick #39: No Gesture Data, but shown by demo programs and i2cdetect

ThePr0gr4mm3r, 2018-06-20:
I have bought the Flick HAT last week, yesterday it arrived. I tested it with a RPi 3 Model B, and it worked fine. Today, the Flick HAT is not giving any data when running the demo, without changing anything.
After some googling, I found a similar problem: https://github.com/PiSupply/Flick/issues/14

Details:
I access the RPi via SSH (Putty).
`i2cdetect -y 1` returns the 42 at the right spot.
`python /home/pi/Flick/bin/flick-demo` starts and shows the expected screen, but does not return any gesture data (the fields on the top stay empty, see linked issue).
I tried using a RPi Zero W, same result.
I also tried to reinstall NOOBS, reinstalled Flick by doing the one line installer, same result.

shawaj, 2018-06-20:
Do you have anything else connected?

Is it getting XYZ data on the demo program?

ThePr0gr4mm3r, 2018-06-20:
The Flick HAT is the only HAT on the Pi. The Pi has ethernet for remote access and (obviously) the power cable connected.

There is no XYZ data available.

ThePr0gr4mm3r, 2018-06-21:
Just made some tests with the RPi 3B and the Flick HAT.

If I unplug the power cable while the Pi is running, it will not work on the next boot and show the already mentioned empty screen (where the data should be). It will work again if i:

- call `sudo poweroff`, wait until the Pi shutted down, pull the power cable and plug in the power cable.
- call `sudo reboot` without plugging.

I hope this helps for further debugging if similar cases occur.

closed 2018-06-21 by ThePr0gr4mm3r

shawaj, 2018-06-21:
Hmm I guess this is some issue due to initialisation of the chip then.

Thanks for the info