PiSupply/Flick #40: Flick large stopped working

OneWhey, 2018-07-03:
Hi!

When I first installed my Flick Large on my Raspberry Pi Model 3 B+ it worked perfectly - for a minute.
Then !without changing anything! I couldn't start the flick-demo any more. So I tried the flick_example in libcpp. This one started but showed a very strange behavior: It detects touches, taps and double taps - very poorly, though. I have to touch and hold for about 2 seconds for it to recognize the touches. There is rarely any gestures recognized and if one is recognized they are shown as "garbage". Touch-recognition works perfectly if I touch any GND spot on my Pi (like the usb or LAN Ports) with my other hand - but gestures still won't work. 

So far I have tried the Flick a new install on my 3 B+ and old and new installs on my 1 B. I also flashed the firmware on each of those. Nothing changed :-/

Is there anything more I can try to do?

ChristopherRush, 2018-07-16:
@OneWhey Picked up your email but going to reply on here since it will be documented for other users. 

Do you have any other HAT boards connected to the Pi at all?

OneWhey, 2018-07-16:
I've had other things connected, no HAT boards, though. Mostly sensors which are connected via I2C.
Disconnecting all of those and only having the flick large connected didn't change anything. Flick still didn't work and showed the same strange behaviour.

OneWhey, 2018-07-18:
@ChristopherRush 
Sorry, I'm new to GitHub. Do I have to mention you in order to have you notified?

ChristopherRush, 2018-07-18:
@OneWhey I'm going to take a look at this now, see if I can replicate this issue. What sort of power supply are you using with the Pi?

OneWhey, 2018-07-18:
@ChristopherRush 
I'm using the original Raspberry Pi Power Supply, so this should not be an issue, right?

This is what I get when trying to run the flick-demo:

`pi@raspi_3:~ $ i2cdetect -yes 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- 5a -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- 76 --                         
pi@raspi_3:~ $ flick-demo 
Did not receive firmware info
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 654, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1441, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 479, in <module>
NameError: name 'sys' is not defined
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
pi@raspi_3:~ $ `

tvoverbeek, 2018-07-18:
What are the other devices on the I2C bus? Flick is on address 0x42, but i2cdetect also sees something at addresses 0x29, 0x5a and 0x76.
From the traceback it seems the RPi cannot communicate with the Flick board at 0x42.
Bad wiring, broken wires ????
Can you control the LEDs on the Flick Large as written in https://github.com/PiSupply/Flick/blob/master/README.md?

OneWhey, 2018-07-18:
@tvoverbeek 
The others are just sensors. They shouldn't have any bad effect on the communication with the flick board. I have already tried the flick board without anything else connected to the i2c bus and it didn't change anything.
I checked the wiring with my multimeter, so this isn't the source of my problem, either. If there would be a problem with my wiring, I shouldn't be able to find the flick's address when scanning with i2cdetect.

Controlling the LEDs is not done via i2c communication, so being able to control those is irrelevant to the main functions of the flick working. But yes, I am able to control them.

ChristopherRush, 2018-08-03:
@OneWhey Can you post a picture of your setup please.

OneWhey, 2018-08-04:
@ChristopherRush 

So I have rebuilt the setup of my Raspberry Pi 3 B+ with only the Flick Large attached to it.
I'm back to being able to run flick-demo but it is very unresponsive to touches and taps unless i ground myself by touching the e.g. LAN port of my Pi. No gestures or air wheel is recognized, though.

I found one more thing, wich is probably a problem: When I run i2detect multiple times, after about thirty times or so, the address of the Flick (0x42) disappears. It won't reappear until i do a reset via gpio. --> `gpio -g mode 17 out; gpio -g write 17 0; gpio -g write 17 1;`
This is without physically touching the setup. Combined with the consistency of the problems, I'm sure it's not about broken wires.

On the backside of the Flick board there is a rogue 10k resistor soldered to the GND/mounting pad of the P1 connector. As it's only connected to GND and only on one side and I compared my board to the ones on the webshop - as much as I could see, the 10k isn't missing anywhere else on the board. So it shouldn't be a problem, but I wanted to mention it.

[image]
[image]
[image]

shawaj, 2018-08-04:
@ChristopherRush have you seen this kind of issue before? I'm thinking maybe a firmware reflash might help but not sure

OneWhey, 2018-08-04:
@shawaj  @ChristopherRush 

I have already reflashed the firmware multiple times

ChristopherRush, 2018-08-06:
@shawaj I have't come across this behaviour before with this board. If there are issues, generally a re-flash would resolve it. 

@OneWhey Do you have any F-F Jumper wires that you could swap over, just so we can eradicate a possible cable issue?

OneWhey, 2018-08-06:
@ChristopherRush I have checked the connections with a multimeter. I have tested them from where the connector on the flick board is soldered to the board across to the pins on the Pi's board. There should not be any issues with the cables.

shawaj, 2018-08-06:
It's worth trying a new cable anyway as it might be frayed internally and be temperamental. Continuity doesn't necessarily mean a good connection

OneWhey, 2018-08-13:
@ChristopherRush @shawaj 
It took a while, but I had to get supplies to make a new cable from my previous workplace.
The flick board is still showing the same strange behavior with the new cable.

shawaj, 2018-08-14:
Sure