PiSupply/Flick #43: When running flicklib demo I get NameError: name 'sys' is not defined

lbilde, 2018-08-14:
Hi Flick :)

When I run flick demo on the pi I keep getting the following error:
pi@raspberrypiLars:~/Flick/bin $ flick-demo
Did not receive firmware info
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 739, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1501, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.5/dist-packages/flicklib-0.0.1-py3.5.egg/EGG-INFO/scripts/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/local/lib/python3.5/dist-packages/flicklib-0.0.1-py3.5.egg/flicklib.py", line 479, in <module>
NameError: name 'sys' is not defined
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/flicklib-0.0.1-py3.5.egg/flicklib.py", line 462, in _exit
  File "/usr/local/lib/python3.5/dist-packages/flicklib-0.0.1-py3.5.egg/flicklib.py", line 330, in _stop_poll
AttributeError: 'NoneType' object has no attribute 'stop'

Any help?

Thanks

lbilde, 2018-08-15:
Hmm seems I can't, using i2cdetect -y 1 results in this:

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

lbilde, 2018-08-15:
Trying to reinstall my pi, can the flick-demo run with python 3.7?

lbilde, 2018-08-15:
After reinstall and using python2.7 I am back and running, thx..

tvoverbeek, 2018-08-15:
@lbilde @ryanteck Yes there is an other issue.
If there is an I2C problem then the atexit function (_exit()) is called.
_exit() stops the polling however at this point the worker thread is not defined yet and hence you get the `AttributeError: 'NoneType' object has no attribute 'stop'` at line 330.
Suppose we have to add a test like `if '_worker' in globals()` before calling _worker.stop()