PiSupply/Flick #46: flicklib demo results in error.

edouin, 2018-09-18:
I am getting this error when trying to run flick-demo.
Can you suggest what to do?

I checked i2cdetect and it shows the device.

Also tried firmware update.

pi@raspberrypi:/usr/src $ flick-demo
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 739, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1501, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 480, in <module>
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 278, in _handle_firmware_info
Exception: An invalid GestiIC Library was stored, or the last update failed
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable

tvoverbeek, 2018-09-18:
@edouin The interesting line in the traceback is:
```
File "build/bdist.linux-armv7l/egg/flicklib.py", line 278, in _handle_firmware_info
Exception: An invalid GestiIC Library was stored, or the last update failed
```
When you look at flicklib.py (https://github.com/PiSupply/Flick/blob/master/flick/flicklib.py)
it seems you were able to communicate with the Flick board, but it did not recognize the correct firmware.
Note there are specific firmwares (different parameters) for each variant (https://github.com/PiSupply/Flick/tree/master/firmware).

Right now I do not have any suggestions, except to try to reload the correct firmware.

edouin, 2018-09-19:
Indeed, flashing solved the GestiIC  library error.
But gesture data is still not registered by the flick-demo.py.
I am using Flick Hat and it was working before.
What can I do for troubleshooting this problem? 

                             **** Flick Demo ****

   ┌────────────────────────────────────────────────────────────────────────┐
   │ X Y Z     :                                                            │
   │ Flick     :                                                            │
   │ Airwheel  :                                                            │
   │ Touch     :                                                            │
   │ Tap       :                                                            │
   │ Doubletap :                                                            │
   └────────────────────────────────────────────────────────────────────────┘

   ┌────────────────────────────────────────────────────────────────────────┐
   │ Firmware valid: Yes                                                    │
   │ Hardware Revison: 99.128                                               │
   │ Params Start Addr: 0x7300                                              │
   │ Library Loader Version: 19.100                                         │
   │ Library Loader Platform: Hillstar                                      │
   │ Firmware Start Addr: 0x1000                                            │
   │ Firmware Version: 1.3.14;p:HillstarV01;x:                              │
   │ DSP: ID9000r2963;i:B;f:22500;nMsg;s:Rel_1_3_prer1784:NM;               │
   └────────────────────────────────────────────────────────────────────────┘

                               Control-C to exit

edouin, 2018-09-19:
Update on the situation.
C++ example ./flick_example is showing the readout.
The problem seams to be in my Python code.

tvoverbeek, 2018-09-19:
@edouin See my answer on Discord

edouin, 2018-09-20:
Still no luck. But getting readout with this library https://github.com/open-electronics/GestIC

edouin, 2018-09-26:
Ended up using direct I2c read write...
...
`bus.write_i2c_block_data(addr, 0x10,[0x00, 0x00, 0xA2, 0x85, 0x00, 0x00, 0x00, 0x7F, 0x00, 0x00, 0x00, 0x7F, 0x00,0x00, 0x00])`
...
`data = bus.read_i2c_block_data(addr, 0x00, 26)  #Read the sensor data payload`
...

This way everything works fine.

ifitikhar-iffy, 2020-04-01:
[image]
I am having the same issue, i don't know how to solve, i've tried reinstalling the OS, i2cdetect is also good, i don't know how flashing solves the issue