PiSupply/Flick #47: No gesture detection

bricolodu13, 2018-10-01:
Hello,

I have the same problem as a few other people here 
I have 3 brand new Flick Hat boards
I have 2 Raspberry Pi 3, one is brand new
I tried with fresh debian and fresh Dietpi
I have plugged the Flick Hat, with : "_i2cdetect -y 1_" none were detected
I have made a cable between the Pi and and the Flick Hat
Now the Flick Hat are now detected with i2cdetect -y 1
when I start the flick-demo 
[image]
Gesture are not detected !
Some here did a fresh Firmware install, I tried it on one of the boards, still the same issue no detection of gesture !

I made a help request on the Flick web site, no answer !!!!

WHAT CAN BE DONE TO CLEAR THIS OUT ?

thanks,
Jean

tvoverbeek, 2018-10-01:
@bricolodu13 Jean,
Strange that i2cdetect shows nothing when the HAT is connected directly, but does show it when you connect with a cable (assume it is a 40 connections GPIO cable, power and i2c only are not enough).
See https://pinout.xyz/pinout/flick_hat.
If you get the firmware info in flick-demo, then at least the first i2c transaction worked.
Two things I can suggest:
1) try to get i2cdetect detecting the HAT when connected directly. Pressed down sufficiently on the pins? Maybe a bad connection for TS (see the pinout diagram).
2) uninstall the flick software (remove everything in usr/lib/python2.7/dist-packages related to Flick and the flick install directory, normally /hone/pi/Flick) and try a fresh install. Note this assumes you are using python2.
Hope you manage to get it going for your demo.

bricolodu13, 2018-10-01:
Hello Ton,

No It is not a 40 connections GPIO cables, here is what is connected 
when I use an external cable between the Pi and the Flick Hat connected as the following maping :
GND          -> Pin 6 (I have checked that all the ground pin on the Flick Hat are connected together)
TS           -> Pin 13
Reset        -> Pin 11
SCL          -> Pin 5
SDA          -> Pin 3
VCC (3V3)        -> Pin 1
5V          -> Pin 2

1/ At home, last week I had one Flick Hat connected directly to the Raspbery Pi 3 with my OS and everything was ok.

My Soundcard goes to the Pi and I have the pins for the Hat as per above

However, I probably made a mistake and when I connected everything, the Raspberry PI 3 get fried.

So I took two other Raspberry Pi 3 (one Brand new) and bought 3 new Flick Hat.
Since then, I couldn't make it working again with Direct connection.
I could only detect I2C and initialize the Flick Hat when connected through the cable.
I suspect a tolerance issue with the 40 pins connectors.

2/ I did a few fresh install of Raspbian and Dietpi but couldn't make them work.
Now, unfortunately, I don't have anymore the time to build a complet new install (I have a lot more then just the Flick)

edouin, 2018-10-02:
Yes, same problem... 
The only solution that worked for me was bypassing the Flick library.
Strange problem, because C++ example and custom Python script works fine.

bricolodu13, 2018-10-02:
Unfortunately, for me, I had my own Python script, but it didnt make it either...

ryanteck, 2020-02-12:
For others to reference: The fix to this is connecting the 3V3 Pin as well on pin 17.

closed 2020-02-12 by ryanteck

vijayrraut, 2021-09-13:
Thanks @ryanteck, connecting 3V3 on pin 17 started working.
I am using flick-zero with RPi-zero and connected headers only for the required pins, and not all the 40 pins.