PiSupply/Flick #49: Communication not working & demos not working after instructrions

narner, 2019-01-25:
Hi there, I'm using a Flick HAT with a Raspberry Pi 3 B+. 

I've followed all the instructions for installing the Flick library, as well as made sure that my i2c interface was configured. 

Running `gpio mode 17 out; gpio write 17 0; gpio write 17 1; i2cdetect -y 1` gives me the following output:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --       
```

And running `flick-demo` gives me the following Firmware info:

```
Did not receive firmware info
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 739, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1501, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 479, in <module>
NameError: name 'sys' is not defined
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "build/bdist.linux-armv7l/egg/flicklib.py", line 462, in _exit
TypeError: 'NoneType' object is not callable

```

shawaj, 2019-01-25:
Did you install wiringpi before loading that?

Is the HAT definitely connected properly to the board? And i2c is enabled?

shawaj, 2019-01-25:
Was there a picture?

narner, 2019-01-25:
Ah there should have been!
[image]