PiSupply/Flick #51: Flick Large + UNO comms

narner, 2019-02-27:
Hi all, 

I'm currently having an issue with the Flick Large + Arduino UNO. I'm using the [Skywriter UNO library](https://github.com/pimoroni/skywriter-hat/tree/master/arduino) for interfacing between the two.

I'm able to get results using the Skywriter board, but not the Flick board... the Arduino sketch I'm trying to use is below, and photos what my wiring setup looks like is attached. Any help would be super appreciated. I had earlier gotten the [same board to work with a Teensy](https://ask.audio/articles/make-your-own-diy-3d-gesture-and-tracking-usb-midi-controller), so I surely should be able to get it working with an Uno...it seems like the callbacks aren't being registered for some reason.

```
#include <Wire.h>
#include <skywriter.h>

unsigned int max_x, max_y, max_z;
unsigned int min_x, min_y, min_z;

// Include string names of gestures/touches for testing
#define SKYWRITER_INC_DEBUG_STRINGS

void setup() {

  Serial.begin(115200);
  while(!Serial){};

  Serial.println("Hello world!");
  
  Skywriter.begin(12, 13);
  
  Skywriter.onTouch(touch);
  Skywriter.onAirwheel(airwheel);
  Skywriter.onGesture(gesture);
  Skywriter.onXYZ(xyz);
}

void loop() {
  Skywriter.poll();
}

void xyz(unsigned int x, unsigned int y, unsigned int z){
  if (x < min_x) min_x = x;
  if (y < min_y) min_y = y;
  if (z < min_z) min_z = z;
  if (x > max_x) max_x = x;
  if (y > max_y) max_y = y;
  if (z > max_z) max_z = z;
  
  char buf[64];
  sprintf(buf, "%05u:%05u:%05u gest:%02u touch:%02u", x, y, z, Skywriter.last_gesture, Skywriter.last_touch);
  Serial.println(buf);
}

void gesture(unsigned char type){
  Serial.println("Got gesture ");
  Serial.print(type,DEC);
  Serial.print('\n');
}

void touch(unsigned char type){
  Serial.println("Got touch ");
  Serial.print(type,DEC);
  Serial.print('\n');
}

void airwheel(int delta){
  Serial.println("Got airwheel ");
  Serial.print(delta);
  Serial.print('\n');
}

```

[image]
[image]
[image]

narner, 2019-03-08:
Hi, has anyone had the chance to look at this yet?

tvoverbeek, 2019-03-08:
@narner I believe the problem is there are no pull-up resistors on the SDA and SCL lines on a UNO.
Teensy has some weak pull-ups.
The i2c-scanner sketch (https://playground.arduino.cc/Main/I2cScanner) does not find anything.
Will try later with some 4k7 pull-ups.

narner, 2019-03-09:
@tvoverbeek sounds good --- just super odd since at one point I did have it working with an Uno...

tvoverbeek, 2019-03-09:
Tried with 4k7 pull-ups. No dice.
Also looked with a scope (Bitscope DSO) to the signals. Look OK both with and without pull-ups, but clock stretching at begin and end.
Maybe something has changed in the Wire library.
So we are back at square one.

tvoverbeek, 2019-03-10:
Have been reading and thinking a bit more.
It seems the MGC3130 on the Flick is not acknowledging any transaction initiated by the 5V side (Uno). It has to pull the SCL line low-high-low as acknowledgment.
There are bidirectional level converters on the Flick board. However for I2C to work with different voltages there need to be pull-up resistors on both the 5V and 3.3V side (the Flick side).
The Flick Large does not have pull-up resistors on the 3.3v side (a design mistake @mmilann ?).
When both sides are at 3.3V it obviously works. Hence using a 3.3V based Arduino works.
Maybe using an external i2c level converter (e.g. https://www.adafruit.com/product/757) between the 5V Uno and the Flick large works.

milajo2, 2021-02-04:
Hi, 

I am also using this flick large but with the arduino mega 2560, I just need the sensor working for the initial stages of my project. However, when I implemented your code using the skywriter library I don't get any output in the serial monitor do I just need to put a 4.7Kohm resistor at the 3.3v? Is it possible you could update me on whether you got the flick working or not?

tvoverbeek, 2021-02-04:
I2C bus needs pull-up resistors on the SDA and SCL lines. The RPi has built-in pull-up resistors for I2C bus 1.
It seems the Mega2560 also has built-in pull-up resistors for I2C on pins 20 (SDA) and 21 (SCL). See the Wire lib docs: https://www.arduino.cc/en/reference/wire

milajo2, 2021-02-04:
Hi, 
I've adjusted my wiring using the SkywriterInterrupt code, whenever I touch the board it doesn't state "Got touch" it just continues on in the serial monitor.