PiSupply/Flick #52: Flick Large communicates but does not recognize gestures

raffaeler, 2019-04-12:
Hello, as it happened to others, I can't make the demo working.
Installation was ok
No other peripherals are connected.
I have no errors and I can turn the green and red len on and off via bash.

I didn't update the firmware. Also, I don't get why the firmware version on GitHub is 1.1 while the board has 1.3. I expected at least the same.

What can I do to make it run?

```
~ $ gpio mode 17 out; gpio write 17 0; gpio write 17 1; i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Running the demo ...
```
                                        **** Flick Demo ****

   ┌────────────────────────────────────────────────────────────────────────────────────────────┐
   │ X Y Z     :                                                                                │
   │ Flick     :                                                                                │
   │ Airwheel  :                                                                                │
   │ Touch     :                                                                                │
   │ Tap       :                                                                                │
   │ Doubletap :                                                                                │
   └────────────────────────────────────────────────────────────────────────────────────────────┘

   ┌────────────────────────────────────────────────────────────────────────────────────────────┐
   │ Firmware valid: Yes                                                                        │
   │ Hardware Revison: 99.128                                                                   │
   │ Params Start Addr: 0x7300                                                                  │
   │ Library Loader Version: 19.100                                                             │
   │ Library Loader Platform: Hillstar                                                          │
   │ Firmware Start Addr: 0x1000                                                                │
   │ Firmware Version: 1.3.14;p:HillstarV01;x:                                                  │
   │ DSP: ID9000r2963;i:B;f:22500;nMsg;s:Rel_1_3_prer1784:NM;                                   │
   └────────────────────────────────────────────────────────────────────────────────────────────┘

                                         Control-C to exit
```

After ctrl-c:
```
$ flick-demo
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 739, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1501, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 147, in <module>

  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 145, in main
```

internet-conception, 2020-11-18:
hello,
Have you figured out what was your problem ? Because I'm at the exact same step 👍
Tks byebye

shawaj, 2020-11-18:
@raffaeler the firmware version of ours is v1.1 buy the "hillstar" firmware for the chip is v1.3. It is a different thing. Sorry for confusion.

@internet-conception are you getting any error message?

internet-conception, 2020-11-18:
hi,
Yes the exact same error has @raffaeler and the same firmware too (hillstar).
How can I use my flick large device ? I should flash the firmware 1.1 from this repo ?
Tks for your help :o)

internet-conception, 2020-11-18:
oh not axactly the same (flicklib V0.0.3)

Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.3', 'flick-demo')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1453, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.3-py2.7.egg/EGG-INFO/scripts/flick-demo", line 147, in <module>
    
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.3-py2.7.egg/EGG-INFO/scripts/flick-demo", line 145, in main

raffaeler, 2020-11-18:
In my case I got a replacement board and it worked.

My guess is that it wasn't really faulty but the software coming with the board is not too basic and not able to clarify the problem, nor to reflash the firmware (that could probably fix the issue).

I still have the old board and it could be helpful to get a 'smarter' software, replace the firmware and see what happens.

FYI I also bought the chips stand-alone (whichi I soldered to one of my boards) and had no problems in making it work. This is the reason why I am very skeptical the chip could be faulty.

internet-conception, 2020-11-18:
Tks @raffaeler 
Have you changed the cable with the new one ? Because it really look like a faulty cable... some reboot the demo dont even start.
I know that I2C is very sensitive about cheap wire...

raffaeler, 2020-11-18:
Of course i did, I have plenty of good-quality dupont-cables and they didn't help at all.
FYI I use many other I2C peripherals and I am very used to them.

My guess is that reflashing the device could help, but I don't want to risk if I don't find a very well documented procedure.

ryanteck, 2020-11-19:
It could be your board is faulty @internet-conception if @raffaeler had the same issue fixed by a replacement board despite it being unlikely could be the case (As if it was a code issue I think we'd have more than just you having the issue and a replacement board wouldn't fix it).

However to try and do a little diagnostics, What Model of Raspberry Pi and Operating system are you using @internet-conception ?

internet-conception, 2020-11-19:
@ryanteck hi I use a genuine pi3b+ with an up to date raspbian.
As I have "nothing to loose" I will try to flash the provided firmware (without compiling the flashing tool I'm not skilled enought for this).
I could not proceed a replacement anyway :p the purchase is far too old now
Tks for your help ! @raffaeler

internet-conception, 2020-11-19:
so cool :) just to let you know that my bord is working !!!
I just flashed the firmware following this instructions : https://github.com/PiSupply/Flick/tree/master/firmware/update
Thanks a lot to you all !

shawaj, 2020-11-19:
Thanks for letting us know. Will close this issue but let us know if we can help with anything else.

closed 2020-11-19 by shawaj

raffaeler, 2020-11-19:
Great to hear that!