PiSupply/Flick #53: Flick demo won't run on Pi3

landrymzb, 2019-05-02:
My flick hat recently just stopped working.It connected to Pi3 perfectly. I ran the i2cdetect and I have 0x42 detected. I even installed the fresh Raspbian, ran an apt-get update, apt-get upgrade then the curl command to install the flick demo code and libraries. 
When I run the flick-demo, and try to use the pad, nothing happens. When I p[ress Control-C to exit the program this is what I see:

-------------------------------------------------------------------------------
flick-demo
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.1', 'flick-demo')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 739, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1501, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 147, in <module>
    
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.1-py2.7.egg/EGG-INFO/scripts/flick-demo", line 145, in main
    
KeyboardInterrupt
------------------------------------
Running flick-demo, I get 
 Firmware valid: Yes                                                                                  
Hardware Revison: 99.128                                                                             
Params Start Addr: 0x7300                                                                            
Library Loader Version: 19.100                                                                       
Library Loader Platform: Hillstar                                                                    
Firmware Start Addr: 0x1000                                                                           
Firmware Version: 1.3.14;p:HillstarV01;x:                                                             
DSP: ID9000r2963;i:B;f:22500;nMsg;s:Rel_1_3_prer1784:NM;

shawaj, 2019-05-02:
@landrymzb that is a normal output when you close the program with a CTRL-C

So what happens when the program is running?

landrymzb, 2019-05-02:
@shawaj the program is running , but I have to execute : curl -sSL https://pisupp.ly/flickcode | sudo bash every time, then flick-demo. Sometimes it runs perfectly, and other time nothing happens.

shawaj, 2019-05-02:
@landrymzb which board do you have?

And when you say sometimes it runs perfectly and other times nothing happens - can you be more specific about what does and doesn't happen?

You shouldn't have to install the software more than once and this could in fact be complicating things further.

Have you tried starting from a new NOOBS / Raspbian install to eliminate any other issues?

landrymzb, 2019-05-02:
It works perfect now on new Noobs. Thanks for your help. I think I started getting complication ever since I started using ssh

shawaj, 2019-05-02:
Perfect, I'll mark this as closed now but if you have any issues in future feel free to post back here

closed 2019-05-02 by shawaj