PiSupply/Flick #54: flicklib.flick returns null when used as annotation

SnowyCoder, 2019-06-09:
If a python user uses an annotation he expects the method to not be deleted, when we use the method as shown in the example the function `flick` used as an annotation returns `None` replacing the method defined by the user with nothing.
The same issue can be found in `touch`, `tap`, `double_tap`, `garbage`, `move` and `airwheel`.
<br>Example:

```python3
import flicklib
@flicklib.flick()
def on_flick(from, to):
  pass

on_flick("north", "south")  # Error
```

Workaround (manual annotation call)
```python3
import flicklib

def on_flick(from, to):
  pass

flicklib.flick()(on_flick)

on_flick("north", "south")  # Success!
```

tvoverbeek, 2019-06-10:
@SnowyCoder That is not how flicklib and the annotations work.
When you import flicklib it already starts a continuous poll of the MGC3130 (the chip on the flick board).
It then reports all possible events to the library.
When you have defined a function with annotation in your own code like
```
import flickllib
@flicklib.flick()
def on_flick(from, to):
    print(from, to)

```
Then the polling loop in the library will call your on_flick function where you can detect which way the flick was (N-S, S-N, E-W, W-E).
What you cannot do is to register a function for only a specific flick.
See e.g. the example flick-snail (https://github.com/PiSupply/Flick/blob/master/bin/flick-snail)

Your workaround works but does nothing.

SnowyCoder, 2019-06-11:
Sorry for the misunderstanding, I'll explain the use case of it:
I have an app that needs to work, no matter what the case is, sometimes the program doesn't receive the input from the flick so my callbacks don't get fired.
To fix it I put another system that calls the callbacks manually using the input from a keyboard.
It works all-right until i try to call my `on_flick` callback, as it has been "eaten" by the annotation that returns `None` instead of the original method.
To work around this I register in the library a copy of the method, so that I can still call the original one.
I'm not sure if this is the intended behavior or not.