PiSupply/Flick #55: flick zero demo error

ifitikhar-iffy, 2020-04-01:
I have raspberry pi model B+ with latest raspbian OS released on 13-02-2020 with flick zero. Flick demo was working fine yesterday but suddenly it stopped working.
Here is the error image
[image]
I reinstalled the OS and flick but the issue is same.
I've tried i2cdetect -q 1, It resuls 42, no problem with hardware.

ryanteck, 2020-04-01:
I can confirm that I can reproduce. I'll start working on if I can fix it.

ChristopherRush, 2020-04-01:
Try updating the Firmware that should fix it.

ifitikhar-iffy, 2020-04-01:
[image]
I don't know how to update firmware but I tried this, but it does not work because Flick is a read only root directory and I don't know how can I change permissions.

ChristopherRush, 2020-04-01:
Add sudo at the beginning of the line

ryanteck, 2020-04-01:
@ifitikhar-iffy I can confirm there was an issue in the library however the firmware update suggested by @ChristopherRush also needs to be done as the error was in relation to telling you there was another error.

If you run the setup line again and then the firmware update (with sudo at the beginning) then it should resolve all issues.

ifitikhar-iffy, 2020-04-01:
[image]
I tried it, I don't know whether I am doing it right or not. copy command replaced the exe file in update directory with enz

ChristopherRush, 2020-04-01:
Follow the instructions here - https://github.com/PiSupply/Flick/tree/master/firmware/update

ryanteck, 2020-04-01:
Hi,

The command run would be incorrect and it looks like by accident you may have deleted a file.

I would suggest the following.

Run command "cd". This should move you back to your "home" folder and just show the pi@raspberrypi line in the console.

Run ```sudo rm -rf Flick/``` this will remove all the flick files that may be badly downloaded.

Run the ```curl -sSL https://pisupp.ly/flickcode | sudo bash``` command again to download them all.

Let it reboot and re-open terminal

cd into Flick and then cd into the firmware folder, then the update folder.

Then run
```sudo cp ../Flick_Zero_V1.1.enz .``` this will copy the firmware into the updating folder.
Then run ```sudo ./flickupdate Flick_Zero_V1.1.enz```

This should update it and fix it.

ifitikhar-iffy, 2020-04-01:
@ryanteck @ChristopherRush A big thanks :) this resolved the issue. fast and great response.

closed 2020-04-01 by ifitikhar-iffy