PiSupply/Flick #56: RPI_REVISION Method of GPIO Breaks flicklib.py When Using RPI Compute Module 3+

MXGray, 2020-07-03:
It's detecting RPI CM3+ as '/dev/i2c-0'  
...  
Instead of '/dev/i2c-1'  
...

ryanteck, 2020-07-03:
Hi @MXGray ,

I'll look into fixing this and should be easy to do. However at the same time this does vary based on what board you are using for the compute module? With the CMIO you'd be wiring the pins directly so could physically change them. But was wondering what host board you're using?

MXGray, 2020-07-04:
Hi @ryanteck ,

Host is <a href="https://www.raspberrypi.org/products/compute-module-3-plus/">Raspberry Pi Compute Module 3+</a>.
It's connected to a <a href="https://docs.luxonis.com/products/bw1097/">Luxonis DepthAI BW1097 board with 3 cameras</a>.
And here's a photo:

[image]

I'm using these components to develop an artificial 3D perception system for blind users like myself.
Except for Flick Zero — I got these as a 1st round pick prize from Luxonis and Intel Movidius for the recently announced <a href="https://opencv.org/opencv-spatial-ai-competition/">OpenCV & Intel Spatial Competition</a>.

And here's a demo video:

https://www.youtube.com/watch?v=mDHOSRhRbsQ

Cheers!  :)

ryanteck, 2020-07-04:
Just had a look on one of my compute modules and I guess the issue is that as the Compute Module doesn't have any fixed GPIO to PIN mapping.

However I suspect most compute modules use the Rev 2 or 3 pinouts so I've changed it to be unless it's a model 1 to try it out. If you try re-installing and report back to me if it's fixed it that'd be great.

tvoverbeek, 2020-07-04:
Looked at the RPi.GPIO source.

For Compute Module 1, 3 and 3+ GPIO.RPI_REVISION returns 0.
For more details use GPIO.RPI_INFO which returns a dictionary. On a Pi4 it returns
`{'P1_REVISION': 3, 'REVISION': 'b03111', 'TYPE': 'Pi 4 Model B', 'MANUFACTURER': 'Sony', 'PROCESSOR': 'BCM2711', 'RAM': '2G'}`
For a compute module 3+ the TYPE returned is "Compute Module 3+"

For flicklib the check GPIO.RPI_REVISION == 1 is OK for using i2c-0 instead of i2c-1.

ryanteck, 2020-07-04:
I've done a commit earlier that does this @tvoverbeek , as the only model of Pi which uses i2c-0 on pins 2 & 3 is the Rev 1 B it made more sense to have it where unless it was that to say that the I2C is used is I2C1 unless it is a Rev 1.

For compute modules it reports 0, for Rev 1 B it reports 1, Rev 2 B it reports 2 and All 40 pins after have been the same so report 3 so it makes more sense to have it do it this way rather than if it matches everything not 1.

tvoverbeek, 2020-07-04:
@ryanteck Agree completely with you.
My comment was meant to provide some background info and was written after I saw your commit.