PiSupply/Flick #57: How-to Request: How to use(Connect) Flick Large with Windows PC

pradeep4748, 2020-07-08:
Hi

Can anyone help on how to connect PiSupply FLick Large(for RPI) , work with Windows PC

tvoverbeek, 2020-07-08:
Flick needs I2C communication. You will need something like an USB to I2C converter with associated device driver.
THen you have to implement the low-level communication with the Flick.
All the details are in the datasheets:  https://github.com/PiSupply/Flick/tree/master/hardware/datasheets

You might have more luck using Windows10-IoT (but then you are back on Raspberry Pi)

ryanteck, 2020-07-08:
The easiest if you wish to use a desktop may be a USB to I2C converter plus then using a Linux distribution of some kind as the code itself would require very minor modifications. But with windows could require a significant amount of work.