PiSupply/Flick #59: Is Python 2.7, need to upgrade to 3

ryanteck, 2020-08-13:
We need to upgrade this to Python 3 at one stage as Python 2.7 is EOL.

tvoverbeek, 2020-08-13:
Yes, also the i2c code needs some refresh.

shawaj, 2020-11-18:
@ryanteck @tvoverbeek could this be the same thing causing #52 ?

ryanteck, 2020-11-19:
@shawaj The customer who first had an issue turned out a replacement board fixed it so it could be a hardware issue.

I don't actually see any Python 2 errors causing #52 as the error message is only given when quitting which doesn't make me think Python 2 is to blame.

Weather the I2C code is causing it I'm not sure, as back in july I did some tweaks to fix the most common error I believe.

I don't think I have a Flick HAT or pHAT so could look at doing the python 3 upgrade if you could send one over (I should only need 1 of the 2 and not both). I think I have the big one but a HAT version would be slightly tidier to work with.