PiSupply/Flick #61: changing pins on flick

milajo2, 2021-02-03:
Hi, 

I was wondering if I can re-assign the pin values to assign elsewhere on the raspberry pi if so what configuration document would I have to change?

shawaj, 2021-02-04:
@milajo2 which Flick board are you using?

milajo2, 2021-02-04:
> @milajo2 which Flick board are you using?
The flick large

milajo2, 2021-02-04:
Also is their possibility of how could use the flick large on Arduino I have an arduino mega 2560

tvoverbeek, 2021-02-04:
Flick large needs I2C (SCL and SDA), 2 GPIOs for Transfer and Reset, VCC, Ground and optionally 2 GPIOs for the Red/Green LED.
You can reassign pins to other GPIOs and I2C (e.g. Pi4 has several I2C possibilities).
Of course you need to modify the flicklib code if you change the I2C bus and any of the other GPIOs.

The Flick uses the same chip as the Pimoroni Skywriter. In their github you will find code for Arduino using the Wire library for I2C. Should work on the Mega2560 which even has 2 I2C buses.
See https://github.com/pimoroni/skywriter-hat/tree/master/arduino

milajo2, 2021-02-04:
Hi, 
I've tried the wire library and it works fine, however now I'm trying to get the serial monitor to output words to indicate gesture and touch. I am using this code, however, nothing is outputting on my serial monitor to indicate that it has found a gesture or touch. 
[image]
[image]
[image]
[image]
[image]
[image]

milajo2, 2021-02-04:
I have to design a project using an rgb LED matrix and this flick board, is there a possibility that I could get this working with the RPI and get my rgb LED matrix working with arduino and combine the two together to get them operating together? As at the moment my LED's aren't working with my RPI and I fully haven't figured out how to get the flick working with the arduino.

milajo2, 2021-02-04:
I've used your advice I'm now getting this message 
[image]

tvoverbeek, 2021-02-04:
On your screenshot the serial monitor is on 9600 baud. In your code above you initialise it to 115200 baud.
Try changing the serial monitor baud setting to 115200.

milajo2, 2021-02-04:
I apologise I now get this.

[image]

milajo2, 2021-02-04:
Also my LED no longer works 😭

milajo2, 2021-02-04:
My flick no longer works with my raspberry pi 
[image]

milajo2, 2021-02-04:
[image]

milajo2, 2021-02-04:
[image]

milajo2, 2021-02-04:
my flick no longer works with my raspberry pi 
[image]
[image]

shawaj, 2021-02-05:
@milajo2 are you sure it was connected to correctly?

It could also be an issue with the firmware