PiSupply/Flick #63: Flick Zero demo doesn't run

Olifiers, 2021-05-31:
Hi folks! I'm having trouble getting my Flick Zero to run, wonder if anyone can help?

I'm trying to run the Flick Zero attached to a Raspberry Pi 4, directly, no cables.  Updated firmware to 1.1. i2cdetect finds it fine (42), but I can't run flick-demo, as I get the error below:

```
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.3', 'flick-demo')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1453, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.3-py2.7.egg/EGG-INFO/scripts/flick-demo", line 147, in <module>
   
  File "/usr/lib/python2.7/curses/wrapper.py", line 43, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python2.7/dist-packages/flicklib-0.0.3-py2.7.egg/EGG-INFO/scripts/flick-demo", line 79, in main
   
_curses.error: addstr() returned ERR
```

Any ideas where should I start?

tvoverbeek, 2021-05-31:
It is a curses error.
Are you by any chance running the demo in a terminal window which is smaller than the standard 80x24?

Olifiers, 2021-05-31:
I am indeed, it's a small display. Zooming out on the terminal sorted the issue, super thanks for the quick solution, sir! ;)

closed 2021-05-31 by Olifiers