PiSupply/Flick #65: Fix for Bullseye 32bit(Solution)

Luciano2018, 2022-02-06:
This is a solution to those have errors in Raspberry Pi OS Bullseye 32 bit, i dont know how to commit this fixxes, so i let here for those  who want fix by yourself.
After the install from curl and the reboot. Delete the scripts installed on `/usr/local/bin `referred to flick. Then copy the ones that are in 
`/home/pi/Flick/flick/build/scripts-3.9/` and fix this 3:

- **flick-2048**(Syntax error):
Line 205 changed to `print (direction)`

- **flick-armcontrol**(Ident error):
Line 50 and 51 to:
```
if delta > 0:
    arm.moveBase(1)
    time.sleep(0.5)
arm.reset()
```

- **flick-sense**(Syntax errors):

Line 272 changed to `print('xyz: ' + xyztxt)`
Line 276 changed to `print('Flick: ' + flicktxt)`
Line 280 changed to `print('Airwheel: ' + airwheeltxt)`
Line 287 changed to `print('Touch: ' + touchtxt)`
Line 291 changed to `print('Tap: ' + taptxt)`
Line 295 changed to `print('Double Tap: ' + doubletaptxt)`

For this last you can use 2to3 to fix:
`sudo apt install 2to3`
then
`sudo 2to3 -w /usr/local/bin/flick-sense`

The others no need update

Luciano2018, 2022-02-06:
And close i, because is only a help for those who are usng this in 2022

closed 2022-02-06 by Luciano2018

tvoverbeek, 2022-02-07:
@Luciano2018 Thanks. Your changes are in github now.