PiSupply/Flick #67: flick-demo error: "TypeError: integer argument expected, got float"

icaruszhu, 2022-02-15:
I have recently acquired a flick large. I have installed relevant code on my Pi 400. 

However, when I tried to run ```flick-demo```, it gives the following error message. 

I would be very grateful to know what went wrong here. (I have been a bit frustrated in order to get the flick large work).  I am not sure there is a hardware problem, but this seems to be a problem related to software. Any advice would be hugely appreciated!

```
Traceback (most recent call last):
  File "/usr/local/bin/flick-demo", line 4, in <module>
    __import__('pkg_resources').run_script('flicklib==0.0.3', 'flick-demo')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 651, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1455, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.9/dist-packages/flicklib-0.0.3-py3.9.egg/EGG-INFO/scripts/flick-demo", line 147, in <module>
  File "/usr/lib/python3.9/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.9/dist-packages/flicklib-0.0.3-py3.9.egg/EGG-INFO/scripts/flick-demo", line 78, in main
TypeError: integer argument expected, got float
```

tvoverbeek, 2022-02-15:
Fixed on github.
In Python 3 you have to use '//' for integer division.
Change '/' to '//' on lines 78 and 79 of flick-demo (or reinstall)

icaruszhu, 2022-02-15:
Very many thanks for such a swift and neat solution! Problem solved--work like a charm now with ```flick-demo```.

It seems that ```flick-snail``` has a similar problem. Would you look into that?

Many thanks again-- I am so happy with flick :-)

tvoverbeek, 2022-02-15:
Thanks for flagging the other occurrence in flick-snail..
Checked the other programs in the bin directory and fixed `flick-armcontrol`, `flick-snail` and `flick-volctrl`

icaruszhu, 2022-02-15:
Thank you again! :+1:

closed 2022-02-15 by icaruszhu