PiSupply/Flick #68: Tap gestures not working well on Pi or Arduino

CJMeek, 2023-09-02:
I am using the flick large and the tap, touch, and double tap gestures do not work very well. Tapping seems to detect multiple taps and a touch, touch seems to detect a touch and a tap, double tap detects all three. Is this a common problem? Does anyone have a solution for this?

danny7894, 2023-09-03:
Try unplugging jumper leads and replugging in, also try reinstall of flickboard code. 🙂

CJMeek, 2023-09-04:
> Try unplugging jumper leads and replugging in, also try reinstall of flickboard code. 🙂

I did try all of this but it’s the same on both my arduino and pi

tvoverbeek, 2023-09-04:
Use the proper mask to only detect the gesture you want.
A tap consists of the sequence touch-finger down-lift finger hence when you enable everything you get multiple gestures.

CJMeek, 2023-09-04:
> 

I have also tried this but it still seems to detect multiple taps when I tap once. When I lift my finger it still keeps going for a little bit.

tvoverbeek, 2023-09-04:
There was an update to flicklib last year due to a Python 3 incompatilbilyt (is_alive instead of isAlive) in Feb 2022.
Also the examples had an update due to the difference for integer division between python3 and python2 (// instead of /) also in Feb 2022