PiSupply/Flick #69: Flick large on Raspberry Pi 5

jmeile, 2024-04-26:
Hi

I'm trying to run this on the Raspberry Pi 5, but as soon as I run the flick-demo, I get:
`RuntimeError: Cannot determine SOC peripheral base address
`

I found that this pi is really different than its predecesors and the main reason is that it isn't compatible with RPi.GPIO (I don't know what this means exactly).

Anyway, I also found that you could solve this by installing lgpio, so, I did it as follows:
```
sudo apt remove python3-rpi.gpio
pip3 install rpi-lgpio
```

The command line demos still doesn't work, so, I tried this code:
https://news.sean.co.uk/2017/11/adding-gesture-control-to-raspberry-pi.html

It started to work, but at some point, an exception will be thrown:
```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/home/admintv/.flick/lib/python3.11/site-packages/flicklib-0.0.3-py3.11.egg/flicklib.py", line 130, in run
  File "/home/admintv/.flick/lib/python3.11/site-packages/flicklib-0.0.3-py3.11.egg/flicklib.py", line 311, in _do_poll
AttributeError: 'NoneType' object has no attribute 'pop'
```

The program will keep running, but the coordinates won't be updated. Exiting with CTRL+C and trying to run it again, just shows:
`lgpio.error: 'GPIO busy'
`

I improve the code a little bit and that "busy" error disappeared, but the "AttributeError" exception still happens. In order to make the code work again, I need to exit from the python interpreter, come back and then run it again.

Is there anyway of getting this to be fixed for the Raspberry Pi 5?

Best regards
Josef

Here the improved version of the code:
```
import RPi.GPIO as GPIO
import os
import flicklib
import time

x_pos, y_pos, z_pos = 0, 0, 0

@flicklib.move()
def move(x, y, z):
  global x_pos, y_pos, z_pos
  x_pos = x
  y_pos = y
  z_pos = z

def print_data():
  while True:
    #os.system('clear')
    print("X:", x_pos)
    print("Y:", y_pos)
    print("Z:", z_pos)
    #time.sleep(0.25)
    time.sleep(0.5)

try:
  print_data()
except KeyboardInterrupt:
  print("Keyboard interrupt, exiting...")
except:
  print("Other error or exception occurred!")
finally:
  GPIO.cleanup()
  print("Cleaning GPIO")
```

jmeile, 2024-04-26:
Ok, I found the source code of flick-demo, so, I copied it to a py file and run it. It runs better, but at some point the exception:
`AttributeError: 'NoneType' object has no attribute 'pop'`
comes.

jmeile, 2024-04-28:
Just for the records, now I understand a little bit the problem. From Raspberry Pi 5, they changed the way how the GPIOs are accessed; older Pis are using some library called RPI.GPIO, which is no longer compatible.

The flick-demo is somehow working, but an error will be raised from the flicklib code. It seems that at some point, no data will be read from the memory and thus the returned object is "None". Here I'm using a python library called: "rpi-lgpio", but you can also use other alternatives:

- lgpio: C libary
- gpiozero: python library using lgpio
- lgpio: python library using lgpio

Currently, I found this link useful:
https://pi-buch.info/gpio-reloaded-i-python

The author explains there how to install and use all those alternatives (only in german, so, you need to speak it or use the chromium based browsers translation).

Coming back to the problem, this is the exception:
```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/home/admintv/.flick/lib/python3.11/site-packages/flicklib-0.0.3-py3.11.egg/flicklib.py", line 130, in run
  File "/home/admintv/.flick/lib/python3.11/site-packages/flicklib-0.0.3-py3.11.egg/flicklib.py", line 311, in _do_poll
AttributeError: 'NoneType' object has no attribute 'pop'
```

So, the error is after this line:
https://github.com/PiSupply/Flick/blob/bada72a01cbec5c44d5635117d3e05736ccdc0a5/flick/flicklib.py#L309

It doesn't read anything and returns "None", so, the next line:
`d_size  = data.pop(0)`

Fails because "data" is "None", so it doesn't have any "pop" attribute. Now, I will try to work with only one gesture and see if the problem comes. My second option is to subclass the flicklib classes and then add a quick an dirty fix, for example:
```
if data != None:
  # Pop out the data from the poll and do whatever you have to do
```

I know it is not a clean solution and it could break something else, but given the fact that I don't know exactly how the code works, it is my fastest way of solving my problem.

tvoverbeek, 2024-04-29:
Remember flicklib was written when only Pi1 and Pi2 existed.
Pi5 is so much faster.
I am also trying to rewrite the lib using only gpiozero since that works on all models.
rpi-lgpio just tries to implement rpi.gpio with the newer libraries.
gpios are only used for the reset line and the transfer line (and the flick large 2-color led)